Audio-Injector/stereo-and-zero #2: What is the maximum voltage range for the Audio Inputs?

Freak7777, 2018-03-28:
I am making an oscilloscope with the stereo soundcard and I want to know what kind of voltages it can handle, and whether I will need to step down the voltages. I am expecting a maximum of around + and - 12v.

flatmax, 2018-03-28:
I believe that the chip handles 1 Vrms - which is about 1.4 V maximum, 
which is 2.8 V peak to peak.

The stereo original card has a resistor divider which halves the input 
voltage, so I would expect that you can input 5.6 V peak to peak with 
the original card.

The zero card has no resistor divider and consequently you should target 
2.8 V peak to peak.

Hope that answers your question - sounds like a great project by the 
way, do you have a url ?

thanks

Matt

Freak7777, 2018-03-28:
thanks for the prompt answer! Looks like I will need to step down the voltage then. No url, I'm just a bored high school student :-P

also, what is the sample rate of the ADC? (I'm assuming the sample rate is determined by the ADC, please let me know if I'm incorrect!) I have a TFT screen on one RPi but it uses almost all of the pins (because it is high resolution and high refresh rate) so I have decided to stream the audio from my sound card RPi through an ethernet cable to the screen RPi and I need to know the audio sample rate.
except maybe the sample rate would be defined in the code I use to stream it... anyway, any help you can give me would be appreciated :-)

Freak7777, 2018-03-28:
actually, thinking about it a little more the sample rate probably has more to do with the operating system. Also, I'm pretty sure the library i am using (SFML) has a function for obtaining the sample rate :-)