Audio-Injector/stereo-and-zero #4: AudioInjector for Zero - low volume of input

ondrejFuchs, 2018-03-31:
Hello,
I have a question about the low volume of recording using the Zero audioinjector. I tried recording on the built-in microphone and the microphone connected to the RCA input and the volume is very low. I use "alsactl -file /usr/share/doc/audioInjector/asound.state.RCA.thru.test restore" or" alsactl --file /usr/share/doc/audioInjector/asound.state.MIC.thru.test restore ". and I did not change anything else. Where might be the problem? The board with RCA input is without potentiometer. I record with "arecord -f S32_LE -c 2 -r 96000 /tmp/test.wav" 

I will be happy for any advice.

flatmax, 2018-04-01:
Can you push an image of your mixer state to the issueImages directory ? ?

thanks

ondrejFuchs, 2018-04-02:
I create Pull Requests (with pictures and 2 records) to main branch. AudioInjector Zero sound card I have attached directly to Zero and nothing else. I set up sound card as you wrote on http://forum.audioinjector.net/viewtopic.php?f=5&t=3&sid=73d0c8938c79732ee5bc52e192fb9a80  (download and install audio.injector.scripts_0.1-1_all.deb.tar.gz, run audioInjector-setup.sh). My last line of /boot/config.txt is 

`dtoverlay=audioinjector-wm8731-audio`

In dmesg i have:
`[   85.922535] audioinjector-stereo soc:sound: ASoC: CPU DAI (null) not registered - will retry
[   85.922567] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[   85.929652] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[   85.929683] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[   87.000012] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[   87.000043] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[   87.057115] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[   87.057144] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[   91.219141] audioinjector-stereo soc:sound: wm8731-hifi <-> 20203000.i2s mapping ok`

First I had a problem recording with a built-in MIC because I had the wrong polarity of the MIC. Now I have it according to http://forum.audioinjector.net/viewtopic.php?f=5&t=3084&start=10 - the bottom picture (microphone connected by wires, because otherwise it did not come to the board). Currently, at my PC's maximum volume, I hear some sound but it is very low.

If you will need another logs from Zero, that will not be a problem. I will be happy for any advice.

flatmax, 2018-04-04:
Your setup looks good.

I can see that the mic. isn't picking much up. Have you tried talking 
close to it or tapping it ?

flatmax, 2018-07-17:
> Did this issue ever get resolved? I have the same problem; I can hook
> the RCA up to a high-volume input but recordings with Audacity or
> arecord are very quiet; same issue with the electret mic.

Hi there,

In the original version of the input breakout board, two resistors were 
switch in manufacture which effected on one channel. This has now been 
fixed.

Can you please contact me off list to organise a replacement for the 
input RCA breakout.

thanks

Matt

skabber, 2020-01-15:
I have the Stereo Soundcard hooked up to a Raspberry Pi Zero.
I have the low volume issue on line in. My line is is coming straight from a turn table. The line in volume nob is turned all the way up. `arecord` produces a very low volume recording. I can test this through both arecord to a wav file or piping to forked-daapd.
`alsamixer` shows the line in volume all the way up.
[image]
Is there something I am missing to get more volume?

Thanks

flatmax, 2020-01-19:
> Screen Shot 2020-01-14 at 11 50 02 PM

Typically a turntable is put through an RIAA phono preamp. This boosts 
the turntable level to a standard line level. The phono preamp also 
shapes the frequency spectrum to make the vinyl audio frequency spectrum 
flat.

skabber, 2020-01-24:
This was totally it. 
Thanks