Audio-Injector/stereo-and-zero #6: Some... quality control issues?

MrShigura, 2018-04-05:
Apologize if I'm misunderstanding something, but... your site shows this...

[image]

[image]

What I got was this...

[image]

No hammer header like the one in the product picture, and with the pin cover that did come with it off, the gpio pins are kind of doing the wave...

[image]

Also one of the rca jacks and the knob next to it are on crooked...

[image]

Again sorry if this was the wrong place to post this, but I didn't see a way to contact you directly and this seems a bit off...

flatmax, 2018-04-05:
> stereoa
> audioinjector pi soundcard side
> img_2121
> img_2135
> img_2133

I understand that whilst the functionality is equal to the picture, 
aesthetic is slightly different.

You will notice that the prototype's horrible through hole capacitors 
have been replaced with surface mount and various other things have 
changed for the better.

If those aesthetics really trouble you, then I encourage you to return 
your item for a refund.

MrShigura, 2018-04-05:
Fair enough, as long as it's as intended, not really an issue, was just making sure before I proceeded, again, sorry for the trouble.