Audio-Injector/stereo-and-zero #7: Makes Kodi lag and no souci outs

lby7980, 2018-04-05:
Hi,
Stereo soundcard works fine on Raspbian Stretch after using the automatic setup. 
Everything goes great using VLC and Audacity. 
Then I've installed Kodi within Raspbian, so I run Kodi as an app in Raspbian. 
In the audio setup of Kodi, the default Alsa is available: pi audioinjector.net. 
But when I use it instead of the HDMI audio output and run a video file, the video completely lags and there is no sound at all. And when I simply exit Kodi, it takes at least 30 seconds to quit and return to the Raspbian desktop, either HDMI audio or audioinjector audio is selected in Kodi. 
Then I used to reinstall Raspbian, then Kodi, without installing the audioinjector stereo soundcard: it worked fine, no lag in Kodi and just a few seconds to exit and return to the desktop. 
After reinstalling the deb file to make the soundcard work, the same problems described above appear again, so I can assume they come from the audioinjector stereo card. 
Any way to resolve this issue ?
Running Raspbian without being able to use Kodi makes unfortunatly this soundcard useless for me... 
Thanks

flatmax, 2018-04-05:
On 06/04/18 02:00, lby7980 wrote:
> Hi,
> Stereo soundcard works fine on Raspbian Stretch after using the
> automatic setup.
> Everything goes great using VLC and Audacity.
> Then I've installed Kodi within Raspbian, so I run Kodi as an app [...]
Hi,

These sound cards don't actually do anything esoteric with the ALSA 
system, they simply load in a different machine driver.

Is it possible that Kodi is using some plugin when you use GPIO sound 
cards ?
Are you using WIFI or ethernet ?

lby7980, 2018-04-06:
Hi,
I do not think that Kodi uses a specific plugin for GPIO hats, as the Hifiberry DAC or the Kuman Pi-Fi DAC work fine with dtoverlay correctly configured. They work fine too with specific OSes like Rune Audio, or OSMC (which is just a Kodi OS based on Debian). 
For ALSA, I cannot confirm because I'm not really experienced with all the Linux subtilities, I can only say that I have to use ALSA mixer to access to Audioinjector Stereo Card "commands", such as enable or disable Hi-pass filter or adjust Capture level, etc.
For the internet connection, I only use Ethernet, I switched off WiFi.