Audio-Injector/stereo-and-zero #11: MCP3008 spi issues?

Molive-0, 2018-04-16:
The stereo audio injector seems to be causing some interference somewhere whilst I am trying to read the output of my two mcp3008 chips. they are both connected via software using the adafruit driver, and are connected to none of the pins the card uses.
whilst trying to debug the issue I tried several things like using hardware spi, using spi1, and using various different pins but the chips only work using software spi without the hat attached.
Can someone help me, as I need the sound card and the ADCs working simultaneously.
I also noticed that there is someone else on the original forum who asked something else about clashes.

Thanks,
Molive.

Molive-0, 2018-04-18:
I'm currently using pins 7 8 9 10 and 11 for spi communitcation over software using spi port 0.
No other pins are used by the chips.

I also have an ads1115 plugged into the i2c, but that's fine with the card and doesn't seem to be the problem.

~Molive

Molive-0, 2018-04-18:
It seems to break even if the audio injector driver isn't loaded, so that seems to lean towards a hardware thing...

KamilPiekutowski, 2020-10-12:
I have been testing Arduino Uno's SPI as a slave. 

Here is the example:

https://roboticsbackend.com/raspberry-pi-master-arduino-uno-slave-spi-communication-with-wiringpi/

 I have similar issue when Audio Injector is connected. When I send message from RPI  it invokes ISR  interrupt callback, but no data can be read on written to SPDR.

Once I unplug the Audio Injector, the connection works as intended. 

Any help?

KamilPiekutowski, 2020-10-12:
I got it to work. Instead of  mounting the  audio injector hat on  Raspberry Pi, I wired everything individually. It is possible something is causing a short on the PCB design of the audio injector. One way to fix it, would be to desolder the header and remove SPI pins completely from the header. Then solder it back knowing SPI pins are not making any connection.

Here is the refference I used to figure out which pins are used by audio injector.
http://forum.audioinjector.net/viewtopic.php@t=4.html#p4

I hope this helps.