Audio-Injector/stereo-and-zero #17: audio injector zero: sound card not found / broken?

soundcamp, 2018-07-07:
I'm trying to get the audio injector zero running with a pi zero.
My audio injector stereo seems to work fine.
With the audio injector zero, when I do:
alsactl --file /usr/share/doc/audioInjector/asound.state.RCA.thru.test restore
I get:
alsactl: load_state:1735: No soundcards found...
And when I try:
audioInjector-test.sh
I get:
sound card not found, is it plugged in correctly ?
is it broken ?
please turn off, fix/change and turn on again.
It's a new soundcard, so no reason to think it's damaged.
Are there other causes I'm missing, I wondered
Thank you for any suggestions
Grant

flatmax, 2018-08-13:
it should just work - is it connected properly ?

joh, 2019-03-22:
I had a similar issue with "no soundcards found". For me it turned out to be a permission problem. My user wasn't in the audio group. Once I added my user to the group audio (```sudo adduser $USER audio```) it worked.

flatmax, 2019-03-22:
good idea.

flatmax, 2019-03-22:
> @flatmax I'm experiencing the exact same issues, brand new audio injector zero, installation works but the device can't be found. My dmesg output says "audioinjector-stereo soc snd_soc_register_card failed (-517)".
> 
> I've tried on 3 different Pi Zero W's one with pre-soldered headers and two where I soldered them and it has the same issue. Also tried Raspian Stretch and Jessie (not via Noobs) and Retropie on all 3 devices in case it was a software issue.

ok - seems like a relevant issue, I'll take this up with you on email

flatmax, 2019-03-22:
> I'm trying to get the audio injector zero running with a pi zero.
> My audio injector stereo seems to work fine.
> With the audio injector zero, when I do:
> alsactl --file [...]

Any luck Grant ?

soundcamp, 2019-03-22:
Hi Matt
I'm afraid this thread got overtaken by other projects.
I d like to get back to it.
Here's a simple question that would be help me:
If I can get the audioinjector working and integrated with my streaming software (which I don't see why I can't), does it permit true stereo inputs from condenser microphones with Plug in Power (if so, what's the bias voltage)?
I'm still not really sure about this and would be really useful to know
Thanks

bw
grant

flatmax, 2019-03-26:
Electret microphones should be fine.

Condenser microphones which require very high phantom power will need a 
phantom power preamp.

Matt

ShawnSmall, 2021-03-02:
Hi Matt,

I too, am having similar difficulties with the audio injector sound zero card. When running `audioInjector-test.sh`, the audio card is not found. The audio card is also not found when I run `aplay -l`. When running `dmesg`, I receive the same error repeatedly that others in this thread have received: `audioinjector-stereo soc:sound snd_soc_register_card_failed (-517)`. I tried `sudo adduser $USER` audio based on Joh's post above, but I was already a group audio user.

I currently have the audio card laying on top of an RPi 4 gpio. I did not want to put solder on the pins or use any connecting clips so that I can disassemble and reassemble my prototype quickly. Do you think it is possible that the audio card is not contacting the gpio pins enough to make a good connection? Or have you encountered any other software fixes to try for this problem with audio injector recognition?

Thank you for any suggestions,
Shawn

flatmax, 2021-03-18:
I would like to remind github users, that we don't officially support here - search for audioinjector people. Use our email list.

Recently we discovered that recent versions of raspbian have a non-robust pulseaudio installation. When we run :
sudo apt remove pulseaudio
The mixer and soundcards work as expected.
Does this fix things for anyone ?

yatbz, 2021-04-24:
Did not work for me. I am still having that snd_soc_register_card failed (-517) error.