Audio-Injector/stereo-and-zero #23: DMA termination issues

jshep321, 2018-12-21:
Hi,

I'm running shairport-sync on an audioinjector (stereo) on a pi zero W.  

Was seeing this issue:
Dec 21 13:14:49 johnnyoffice kernel: [  203.973061] bcm2835-i2s 20203000.i2s: I2S SYNC error!
Dec 21 13:14:49 johnnyoffice kernel: [  203.993806] bcm2835-i2s 20203000.i2s: I2S SYNC error!
Dec 21 13:14:50 johnnyoffice kernel: [  204.633187] bcm2835-dma 20007000.dma: DMA transfer could not be terminated
Dec 21 13:14:50 johnnyoffice kernel: [  204.634273] bcm2835-i2s 20203000.i2s: I2S SYNC error!

The problem here is that the DMA transfer to the card won't work, because the output is muted by default.  To fix, I did this:  Run alsamixer and unmute (press M) on the "Output Mixer HiFi".   Then run sudo alsactl store to enable these settings post-reboot.  

This should REALLY be included in the setup instructions (or install script) as this is certainly not obvious.

My theory: dma termination won't work if there is no endpoint.  And that endpoint, for some reason beyond my understanding, doesn't exist/enumerate/etc. until the target output is UN-muted.

Additional insight welcome!

twofingerrightclick, 2020-04-02:
This gave me a good place to start as I have the same `DMA transfer could not be terminated` message. But I still have the message after unmuting `Output Mixer HiFi`

flatmax, 2020-04-02:
DMA transfer is a very low level Kernel thing. DMA is common for all I2S 
devices.

Agreed, setting the output Mixer HiFi to on should definitely be 
included in the setup instructions.

Not sure the two are related however.

==================================================
Our latest product launching now on Kickstarter :
https://bit.ly/2Qr1557

jshep321, 2020-04-05:
@twofingerrightclick Does the audio work and you're seeing this, or seeing the message and still no audio?  I can use my working setup (used daily) to help debug if you like.

twofingerrightclick, 2020-04-06:
@jshep321 That is awfully nice of you. No the audio doesn't work. When I test using `aplay sound.wav` it makes a little static sound briefly at the start of playing the sound clip. When the clip is over I get the `DMA transfer` error. I am using this kernel: 
- Linux samplerbox 4.14.30-v7+ #1102 SMP 
I got the image here http://homspace.xs4all.nl/homspace/samplerbox/SBbuild.html

Do you think I need a different kernel?

jshep321, 2020-04-18:
Yeah, you likely have some conflict here.  Try setting it up on the default Raspbian (following the audioinjector instructions and my suggestion above) and I suspect your problems will be resolved.