Audio-Injector/stereo-and-zero #24: need to get working with cli-visualizer

djloot, 2019-01-30:
I assume this will require fifo of some type but can't figure out how to get the slave.pcm defined. 
here is the section in the documentation for the system:

ALSA setup config

pcm.!default {
    type file               # File PCM
    slave.pcm "hw:0,0"      # This should match the playback device at /proc/asound/devices
    file "|safe_fifo /tmp/audio" #safe_fifo will be in the cli-visualizer/bin/safe_fifo directory
    format raw              # File format ("raw" or "wav")
    perm 0666               # Output file permission (octal, def. 0600)
}

flatmax, 2019-01-31:
Your idea looks good. I don't have an example for piping to fifo off hand.

Out of interest, does the audio have to go to file/fifo ? Can you pipe 
directly in to your cli-visualizer ?

djloot, 2019-01-31:
It doesn't have to go fifo, just don't know enough about the code to set it
to go directly to the alsamixer.  It says to to match /proc/asound/devices.
For me, it looks like this:
 0: [ 0]   : control
 16: [ 0- 0]: digital audio playback
 24: [ 0- 0]: digital audio capture
 32: [ 1]   : control
 33:        : timer
 48: [ 1- 0]: digital audio playback
 49: [ 1- 1]: digital audio playback
 56: [ 1- 0]: digital audio capture
 57: [ 1- 1]: digital audio capture

djloot, 2019-02-01:
I honestly don't know. It looks like it was developed kind of based on CAVA.  I got CAVA working perfectly with the sound card but also wanted cli-visualizer.