Audio-Injector/stereo-and-zero #25: Is it possible to use Audio injecter input and HDMI Output?

nospam2k, 2019-01-31:
I want to be able to use Audio-injector input and feed directly to HDMI audio output. Is this possible? If so how?

flatmax, 2019-01-31:
Yes,

You need to arecord in from the audio injector and aplay out to the HDMI 
device.

I am not sure how your devices are set up. Imagine your Audio Injector 
card is device 0 and the HDMI is device 1 :

arecord -f S32_LE -c2 -r 48000 -D hw:0 | aplay -D hw:1 -f S32_LE -c2 -r 
48000

If your HDMI requires a different sample rate, format or channel count 
then try using plughw instead (also if you get terrible sample drops) :

arecord -f S32_LE -c2 -r 48000 -D hw:0 | aplay -D plughw:1

Hope that helps !

Matt

nospam2k, 2019-01-31:
Thanks for the feedback! The real problem is I don't know how to get them to exist together since I have to comment out dtparam=audio=on in order to add dtoverlay=audioinjector-wm8731-audio.

So my real question is: how do I configure for both hdmi and audio injector to coexist?

nospam2k, 2019-02-01:
Ok, I think I'm good. After hours of playing (ugh) I found I didn't need to comment out dtparam=audio=on and just could use alsaloop!

closed 2019-02-01 by nospam2k