Audio-Injector/stereo-and-zero #27: Need some installation help with zero card

Kman91, 2019-02-15:
I have installed the zero card on my 3B+.  I am kind of a newbie, and I had a hard time installing the .deb file, but it finally seemed to work.

I ran the installation script and it said it was installed, and I rebooted, but it won't do the test or output any sound.  It says "Couldn't find the sox command please install the sox package" when I try to run audioInjector-test.sh.

I don't know what to do now.  I'm using this with OpenAuto pro as a head unit in my car.  It works great with the USB plug in sound card, but not the Zero yet.

Thanks!

Kman91, 2019-02-22:
I've posted this for several weeks now on both here and on the Audioinjector site, no help.  I don't know what to do here, but it's just not working and I'm stuck.  

If I don't get some help soon I will have to send this back and report with feedback on Amazon (where the feedback is already not great).  I really don't want to leave negative feedback for what looks like a really nice little sound card, but I don't know what else to do.

flatmax, 2019-02-28:
To install sox :

sudo apt install sox

Anyother applications you want to use ?

Kman91, 2019-02-28:
Thank you for this help.  

I'm not specifically trying to install any applications, I'm just trying to get it to work.  

I think I installed the .deb file after wondering floundering around for a while trying to figure out what that meant.  I don't know if I could do it again, because I'm not even sure what I did, but in the end it said something about files being installed.

When I run the test it said sox wasn't installed (as I stated above).  I hope once I install this it will work.

I hate to be critical, but the installation instructions leave a lot to be desired for newbies like me.  I've been able to get things working pretty well on the Pi other than this, but I've never installed drivers or anything yet.

Thanks again for the help.

Kman91, 2019-03-01:
It started to work, but didn't.

It says it will required additional disk space and asks me if I want to proceed, and I say "y".  Then it looks like it's trying to download 4 things from raspian.raspberrypi.org, but in all 4 cases I get a 404 file not found error.  It is connected to the internet over wifi at the time and it downloaded the install package just fine, so it's connected.  

I tried this twice, with the same results each time.  It also doesn't recognize many of the commands like asound at this point.

Any ideas?

Kman91, 2019-03-01:
I also notice when I type aplay -l, it says "aplay: device_list:270: no soundcards found...

I did the audioInjector-setup.sh again while connected to the internet and it ran fine (again), but after 
the requested reboot, still seems broken.

flatmax, 2019-03-04:
ok - it doesn't see your sound card.

Is the card plugged in correctly ? Which card do you have ?
What does your /boot/config.txt file have in it ?
What does your dmesg command say ?

Kman91, 2019-03-06:
The Zero sound card is connected to female headers that plug into the raspberry pi (3B+) headers, soldered so the components are on the top side, just like the pi.  It's soldered well.  I don't know what could be a problem there.  There are also pins I've soldered to the top for the 7" Pi display, but only 4.  Power and I2C (first 4 pins only).

Here are the config.txt (a screen dump) and dmesg screen dumps (attached)
[dmesg.txt](https://github.com/Audio-Injector/stereo-and-zero/files/2937655/dmesg.txt)
[config.txt](https://github.com/Audio-Injector/stereo-and-zero/files/2937656/config.txt)

Thanks for the help!  I hope I can get this working, I'd really like google voice input.

Kman91, 2019-03-20:
I'd really like to get this working.

yatbz, 2021-04-24:
I think you have to activate "#dtoverlay=audioinjector-wm8731-audio" in config.txt by removing the "#".