Audio-Injector/stereo-and-zero #28: In line in input left channel is picking up the signal but right channel is not picking up.i have used the same asound.state.RCA.thru.test. any changes to be done in that file ??(seperate control structure for left line in and right line in for input mux)

saimaheshh, 2019-02-28:


saimaheshh, 2019-03-01:
Please help

saimaheshh, 2019-03-03:
Please help..we are in the middle of a project and it got stuck

flatmax, 2019-03-04:
Can you please type your message here - what do you need help with ?

saimaheshh, 2019-03-04:
Sir/madam
1.i have configured the audio card with procedure u have mentioned and
files given.
2.i have soldered a female jack to the input (L ,R,G) of audio card which
takes the input.
3.i m recording with the help of arecord command with 2 channels.
4. I have given the input to audio card  with the use of function generator.

Problem
1.left channel is picking up the signal but right channel is not picking
up.i have checked this with audacity software running on the other system.
2.input to the audio card seems fine as I checked that with an oscilloscope

Please help me with this.

saimaheshh, 2019-03-06:
Plzz help