Audio-Injector/stereo-and-zero #30: using mic and line input together

azardiego, 2019-07-14:
Hi, I was wandering if there is a way to use one line input and the onboard mic together in the audioinjector stereo card?

I have a raspberry pi 3b+ with de audio injector stereo card perfectly instaled and working fine. In the alsamixer there is a selector to select between de two inputs so I supose I can't get that task done. But perhaps it's a driver issue and there's a solution. 
I installed the card with the standar procedure with the scripts given in these package: audio.injector.scripts_0.1-1_all.deb.tar.gz
Thanks!

flatmax, 2019-07-15:
Good to hear it is all working!

Inside the card, there is a multiplexer which selects between the line 
input or the microphone input. If you look at the datasheet here at the 
image on the first page :

https://statics.cirrus.com/pubs/proDatasheet/WM8731_v4.9.pdf

In the image is the "MUX" block which selects between the microphone and 
the line inputs. Unfortunately the mux is like a switch - not a mixer. 
For that reason, you can't record from both simultaneously.

Matt

azardiego, 2019-07-15:
Hi Matt, thanks for the fast response. The diagram is very clear. I needed to capture one line with "hi fi" audio and wanted to use the microphone to recieving audio data in the mid range without any concerns about dinamic range or "hi fi"  because it will be used to analyze certain parameters, I thought that would have been perfect. I will have to put a mic pre in the other line then,

Thanks again!