Audio-Injector/stereo-and-zero #31: Stereo Card for SIP calls

drfrutuoso, 2019-07-24:
Hi,

I just received the sound card today and so far I tested the headphone jack and MIC recording, both fully functional! That being said, as the title suggests, I want to use it so I can have a SIP server/client on a RPi 3 and for this I'm using Twinkle (fully tested before with a USB mic and headphones). 

With the sound card, whenever I start Twinkle I get "Critical: Opening ALSA driver failed: snd_pcm_hw_params failed: Invalid argument" ... any ideas on how to solve this?

The sound card even shows up on Twinkle settings as "ALSA: plughw:0,0: audioinjector-pi-soundcard ()" so I'd assume everything seems fine.

Any help is appreciated.

flatmax, 2019-07-24:
I also would assume that everything would be fine because you are using 
the plughw device. All incompatable sample rates and bit depths would be 
converted for you in software ... hmmm

Any ideas what "argument" it is talking about ?

drfrutuoso, 2019-07-25:
On a side note, might not be related since the sound card is indeed working, but on dmesg I'm seeing the following (see image attached). I'm intrigued with the supplies not being found statements?!

Regarding the arguments I've honestly no idea. Also when I make a call I'm able to hear the ringtone but as soon as I answer it tells me that it cannot access both speaker and microphone.

[image]

drfrutuoso, 2019-07-25:
I managed to make it work, not entirely sure how but what I did was install pulseaudio related packages and with that Twinkle now assumes the speaker as ALSA: default and the mic as ALSA: plughw:0,0

If you have any idea as to why I'm seeing those missing supplies statements please let me know. Hopefully I'll be able to test out the RCA I/O today also :)

flatmax, 2019-07-26:
> 2019-07-25-103513_1920x1080_scrot

The dummy regulator lines are not important. Linux allows you to delay 
until regulators are stable - however I don't see any significance in 
this as these regulators are stable from early on in the boot cycle. 
Thus it isn't important to have non-dummy regulators.

flatmax, 2019-07-26:
ok - sounds like a twinkle configuration problem. Glad to hear you got 
it to work !

drfrutuoso, 2019-08-02:
Allow me to "re-open" my previous issue to ask something that maybe you'll be able to provide some help. Apologies if it is too much out of context.

So I managed to have everything functional, even adding a class D stereo amplifier (TPA3110D2PWP)  on a custom made PCB I had develop before. My issue is the feedback noise I get from having the mic close to the loudspeaker. I've been experimenting with the Speex codec which has some form of AEC algorithm but with no noticeable difference. So I'd like to ask if there's any form of additional hardware or software you could recommend for further testing?

Thanks in advance!

flatmax, 2019-08-06:
Good going !
It is a little tricky having a microphone right next to a speaker. I am not sure of software which will do feedback cancellation. There are such algorithms. If you find a good one, please let us know.