Audio-Injector/stereo-and-zero #32: Stereo sound card does not seem to be recognized with Raspbian Buster

nls-GitHub, 2019-09-23:
The stereo sound card does not seem to get recognized with the newly released Raspbian Buster - https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/

flatmax, 2019-09-23:
Any ideas why ?

Does it install ? Any info in "dmesg" ?

Can you see it when you do aplay -l ?

nls-GitHub, 2019-09-23:
Sorry, I do not know why, except perhaps some incompatibility with the new
kernel?  The board would not show up with the aplay -l, I did not check
dmesg.  The board is in use elsewhere, I had to revert back to Stretch, so
I cannot experiment.  Do you have a test setup that you can verify with?

vasilisvg, 2019-10-19:
The Stereo sound card works fine on my raspberry pi with Buster Lite installed.

nls-GitHub, 2019-10-19:
Is this with the 2019-09-26 version - https://www.raspberrypi.org/downloads/raspbian/?  I had tried with the earlier 2019-07-10 version and could not get the card to be recognized, so perhaps I will try again now.   Thanks.

vasilisvg, 2019-10-19:
Yes indeed, with the latest release with everything up to date.

jongrady, 2019-11-18:
I'm having issues getting the Stereo card to work with buster.  It appears to be recognized and I can see it with aplay -l but I am unable to get sound out of the 3.5mm jack on the board.

I checked alsamixer and made sure the settings were correct.  

Here is the output of my dmesg:

[    5.140731] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[    5.140754] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    5.150521] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    5.162480] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.162757] usbcore: registered new interface driver brcmfmac
[    5.163600] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[    5.163617] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    5.186294] Console: switching to colour frame buffer device 100x30
[    5.216313] vc4-drm soc:gpu: fb0: DRM emulated frame buffer device
[    5.283083] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[    5.283105] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    5.513626] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    5.513729] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    5.514920] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[    6.262438] wm8731 1-001a: Assuming static MCLK
[    6.262472] wm8731 1-001a: 1-001a supply AVDD not found, using dummy regulator
[    6.262590] wm8731 1-001a: Linked as a consumer to regulator.0
[    6.262607] wm8731 1-001a: 1-001a supply HPVDD not found, using dummy regulator
[    6.262672] wm8731 1-001a: 1-001a supply DCVDD not found, using dummy regulator
[    6.262741] wm8731 1-001a: 1-001a supply DBVDD not found, using dummy regulator
[    6.269066] audioinjector-stereo soc:sound: wm8731-hifi <-> 3f203000.i2s mapping ok

jongrady, 2019-11-22:
@flatmax Any ideas on my comment above?

flatmax, 2019-11-24:
Is the mixer in a known good state ?

When you open alsa mixer, is the 'Output Mixer HiFi Playback Switch' 
enabled ?

  ![This image shows the mixer 
settings](https://raw.githubusercontent.com/Audio-Injector/stereo-and-zero/master/issueImages/wm8731.workingOutputMixer.png)

Matt

jongrady, 2019-11-25:
I'm not 100% sure I know what you mean by "known good state".  I will say that it was working fine on a previous version of Raspbian using all the same hardware and only stopped after flashing the SD Card to Buster.

The Output Mixer Hifi Playback Switch is enabled in alsa mixer.

jongrady, 2019-12-02:
@flatmax Is there any other information I can provide to help troubleshoot this?  I've been pulling my hair out trying everything I can think of to get it to work and keep coming up empty.

flatmax, 2019-12-02:
OK - I'll make some time ASAP to try with the new version of Buster Raspbian.
From what you have said, the card is recognised by the kernel, but there is no sound.
You are plugging into the jacks on the Audio Injector card and using the .deb file to install.

jongrady, 2019-12-02:
Correct.  Let me know if you need any more info from me or want me to try anything.

jongrady, 2019-12-09:
@flatmax Any luck?

jongrady, 2019-12-23:
@flatmax I upgraded to the latest Raspbian and am still not able to get any sound from the headphone jack.  Any help or suggestions is appreciated.  Thanks!

nls-GitHub, 2020-01-12:
Have you setup alsamixer so that the headphones are connectd to Line In?
[image]

flatmax, 2020-01-12:
Not aware of that setting. Generally the headphones are a mirror of the output from the DAC. Have a look at the block diagram of the chip here :
https://statics.cirrus.com/pubs/proDatasheet/WM8731_v4.9.pdf

nls-GitHub, 2020-01-12:
My application for the board is recording, and I had to set it to monitor the input data on headphones,  Perhaps it is not necessary for output.