Audio-Injector/stereo-and-zero #33: Adding a power button

vasilisvg, 2019-10-05:
I’d like to add a power button to the phono streamer I am making. [On this site I found a rather simple and I think elegant way to that](https://howchoo.com/g/mwnlytk3zmm/how-to-add-a-power-button-to-your-raspberry-pi): by shorting GPIO pins 5 and 6 you can easily halt and resume the Pi. 
[In this issue about GPIO pin layout](https://github.com/Audio-Injector/stereo-and-zero/issues/18) I read that pins 5 and all ground pins are used by the Stereo card. 
Now I am wondering: will this simple power on/off trick still work if I add in on top of the Stereo? Or will it not work and (possibly) damage the card?

flatmax, 2019-10-09:
Gpio 6 is a common ground. You should be fine to use that. GPIO 5 is a 
common I2C clock line. If it works, you can use that as well.

Should be fine to test it.

Matt

vasilisvg, 2019-10-18:
It works, but not automatically. I am piping input from the phono pre-amp to forked-daapd. If I run the `listen-for-shutdown.py` script *before* I start the pipe, it doesn't work: the pipe breaks after a few seconds. So the order of starting things up is important here. 
In the contab I run two scripts at `@reboot`. The first one simply tells the pipe to start:

```
#! /bin/sh
arecord -c 2 -f S16_LE -r 44100 > /home/user/music/pickupPipe
```

And the other starts with a pause of ten seconds, and then starts the script

```
#! /bin/sh
sleep 10
sudo python /usr/local/bin/listen-for-shutdown.py
```

This is the only way I got it to work.

flatmax, 2019-10-22:
Good stuff.

Another way to start processes in a particular order is to use the init 
scripts. Are you aware of the init scripts ?

vasilisvg, 2019-10-23:
I have read about them, and I should probably take a closer look at them.