Audio-Injector/stereo-and-zero #35: High resolution piping issues

vasilisvg, 2019-10-19:
Yesterday I received my Stereo sound card.. I want to use it to pipe the input from a phono pre-amp to [forked daapd](https://github.com/ejurgensen/forked-daapd/). This works fine when I use 

```
arecord -f S16_LE -r 44100 -c 2 -t wav > /home/pi/music/pipe
```

Forked daapd automatically picks up the pipe and plays it. This turned out to be much easier that I thought (-:

But when I try to get a higher resolution problems occur. For instance when I try to use S24_LE, I get continuous `"overrun!!! (at least 124.002 ms long)"` warnings. And it sounds like the world is exploding. I tried adding some buffer on the arecord side of the pipe (`-B 4000`, or `-B -4000`, not sure how this works exactly) but that doesn't seem to do much.

Now I am wondering where to look for a solution. Is it a write speed issue? Do I need a microSD card with faster write speed than the SanDisk Ultra 16GB I am using right now? I guess write speed doesn't count with a pipe. Or is it a processor thing? I run the Stereo card on a Raspberry Pi 3 Model B Plus Rev 1.3 with an up to date Raspbian Buster Lite. Would a Raspberry Pi 4 solve these issues? Should I try overclocking?

Or do I need to look in a completely different direction? In the end I would love to use a resolution of -f S24_LE -r 96000, but right now this seems way out of reach. Any tips and help are very much appreciated!

flatmax, 2019-10-22:
What happens when you use S32_LE as the resolution ?

It is possible there is less CPU operations with S32_LE.

vasilisvg, 2019-10-23:
It seems to be an issue with forked-daapd after all. [Things are being fixed over there now](https://github.com/ejurgensen/forked-daapd/issues/830)