Audio-Injector/stereo-and-zero #36: Audio Server Web Access

Pijit, 2019-10-30:
It looks like the stream is working, but I can't access it via a web page...

(Started with a command in Terminal)
(ffmpeg -f alsa -i hw:0,0 -acodec mp3 -f rtp rtp://192.168.0.22:24)

TERMINAL DISPLAY...

Input #0, alsa, from 'hw:0,0':
  Duration: N/A, start: 1572378765.640398, bitrate: 1536 kb/s
    Stream #0:0: Audio: pcm_s16le, 48000 Hz, stereo, s16, 1536 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (pcm_s16le (native) -> mp3 (libmp3lame))
Press [q] to stop, [?] for help
Output #0, rtp, to 'rtp://192.168.0.22:24':
  Metadata:
    encoder         : Lavf58.20.100
    Stream #0:0: Audio: mp3 (libmp3lame), 48000 Hz, stereo, s16p
    Metadata:
      encoder         : Lavc58.35.100 libmp3lame
SDP:
v=0
o=- 0 0 IN IP4 127.0.0.1
s=No Name
c=IN IP4 192.168.0.22
t=0 0
a=tool:libavformat 58.20.100
m=audio 24 RTP/AVP 14

[alsa @ 0x17d51e0] ALSA buffer xrun.rate= 128.8kbits/s speed=0.501x    
[alsa @ 0x17d51e0] ALSA buffer xrun.rate=  87.7kbits/s speed=0.764x 
...

END TERMINAL DISPLAY

HTML code in my web page...

<P><H3>Live Stream</H3>
<audio controls>
<source src=”rtp://192.168.0.22:24" type=”audio/mp3”>
</audio>

I've tried variations on this code, but nothing works. Any ideas???
Thanks

flatmax, 2019-10-30:
The HTML code didn't come through!

Pijit, 2019-10-30:
Yeah, I noticed that when I went back to look at the post today. Fixed. Take another look. 

-Pete

flatmax, 2019-10-30:
It is possible you want to look into setting up some sort of casting 
server. For example, this article uses icecast :

http://www.streamingmediaglobal.com/Articles/Editorial/Featured-Articles/DIY-Live-Audio-Streaming-Using-Icecast-with-FFmpeg-125665.aspx

Pijit, 2019-10-31:
Tried Icecast before, trying again... getting this error...

Terminal command to start stream:
ffmpeg -f alsa -i hw:0,0 -acodec libmp3lame -ab 32k -content_type audio/mpeg -f mp3 icecast://source:source@192.168.0.22:8000/stream

result:
HTTP error 401 Authentication Required
icecast://source:source@192.168.0.22:8000/stream: Server returned 401 Unauthorized (authorization failed)

icecast.xml:
<authentication>
        
        <source-password>icecast</source-password>

flatmax, 2019-10-31:
In the past I have used ffserver however I am not sure if that would 
play through an html5 audio element. However it may be worth a try.

I seem to remember launching it like so :

ffserver -f /etc/ffserver.conf

Here are some pointer ffmpeg calls I would use for this ... not sure 
which one actually worked (probably the localhost one).

ffmpeg -f alsa -ac 2 -ar 48000 -i hw:0,0 -acodec pcm_s32le -f ffm 
-acodec wav http://192.168.0.5:8090/inputAudio.wav
ffmpeg -f alsa -ac 2 -ar 48000 -i hw:0,0 -acodec pcm_s16le -f ffm 
http://127.0.0.1:8090/inputAudio.ffm

Here is a sample ffserver.conf file :

################ ffserver.conf ######################

HTTPPort 8090
HTTPBindAddress 0.0.0.0
MaxHTTPConnections 2000
MaxClients 1000
MaxBandwidth 15000

# Access log file (uses standard Apache log file format)
# '-' is the standard output.
CustomLog -

##################################################################

<Feed inputAudio.ffm>
File /tmp/inputAudio.ffm
FileMaxSize 200K
#Launch ffmpeg
# Only allow connections from localhost to the feed.
ACL allow 127.0.0.1 192.168.0.100
</Feed>

##################################################################

<Stream inputAudio.wav>

# coming from live feed 'feed1'
Feed inputAudio.ffm
Format wav
AudioChannels 2
AudioSampleRate 48000
AudioBitRate 3072
NoVideo
AVOptionAudio flags +global_header
#Preroll 1

</Stream>

##################################################################
# Special streams

# Server status

<Stream stat.html>
Format status

# Only allow local people to get the status
ACL allow localhost
#ACL allow 192.168.0.0 192.168.255.255

#FaviconURL http://pond1.gladstonefamily.net:8080/favicon.ico
</Stream>

##############################################

Pijit, 2019-11-28:
I managed to get IceCast working with access via a web page using FFMPEG, but it skips and the sound quality is very thin. I tried to use ICES, but it won't play any sound from the same web link. The log seems to indicate that everything is starting up OK...

[2019-11-26  15:02:06] INFO ices-core/main IceS 2.0.2 started...
[2019-11-26  15:02:06] INFO input-alsa/alsa_open_module Opened audio device hw:0,0
[2019-11-26  15:02:06] INFO input-alsa/alsa_open_module using 2 channel(s), 44100 Hz, buffer 500 ms 
[2019-11-26  15:02:06] INFO input-alsa/alsa_open_module Starting metadata update thread
[2019-11-26  15:02:06] INFO signals/signal_usr1_handler Metadata update requested
[2019-11-26  15:02:06] DBUG metadata/metadata_thread_signal meta thread wakeup
[2019-11-26  15:02:06] DBUG metadata/metadata_thread_signal reading metadata from "/home/pi/ices-2.0.2/src"
[2019-11-26  15:02:06] INFO encode/encode_initialise Encoder initialising in VBR mode: 2 channels, 44100 Hz, nominal 64000
[2019-11-26  15:02:06] INFO metadata/metadata_thread_signal No metadata has been read
[2019-11-26  15:02:07] INFO stream/ices_instance_stream Connected to server: 192.168.0.25:8000/stream

My HTML code to access the stream looks like this...
<audio controls>
<source src="http://192.168.0.25:8000/stream">
</audio>

The IP address and stream name is matched in the related XML files.

Pijit, 2019-12-29:
I must be very close. No errors except on the webpage in the progress section of the audio playback bar.  I do get this error if I try to use VLC Player to play the audio; "VLC is unable to open the MRL 'v4l2://'. Check the log for details. Media Resource Locator? V4L2://? What's that?

Pijit, 2020-01-04:
V4L2 appears to be a video source. I don’t see a setting in VLC to capture only audio. This would be a nice way to test the audio card and maybe even stream. I like the GUI. 
[image]