Audio-Injector/stereo-and-zero #38: is the stereo sound card compatible with Raspberry Pi 4?

smomid, 2019-12-03:
Hi, I am curious to try the stereo sound card with a Raspberry Pi 4. Is it compatible with Rpi 4?

Thank you.

flatmax, 2019-12-03:
Yes it is.

The I2S silicon on the Pi4 and Pi3 are the same. This means that the 
same audio drivers will work across Pi versions.

smomid, 2019-12-04:
Awesome thank you. 

Nick

w4mmp, 2020-01-06:
Hello,
Just installed the zero sound card on a RPi 3 running the most current version of Raspian (Buster).  The install script was run and that appeared to work.  However when running sudo aplay -l, it reports "No soundcards found... ".   I am at a loss where to look.

flatmax, 2020-01-11:
OK - what if you do a manual setup ?

I'll have to make some time to look into this.

w4mmp, 2020-01-15:
I determined the cause.   Both the I2C and I2S modules need enabled.   I had I2C enabled but not I2S.  It appears that with Buster, these disabled by default.

flatmax, 2020-01-19:
Thanks for working that out.

montsamu, 2021-01-11:
In addition to I2C and I2S modules needing enabled, for Buster the syntax for activating the the HiFi mixer with the amixer command was slightly different for me. I needed to specify "-c 0" as part of the command to select card 0, even though it was the only and default card.