Audio-Injector/stereo-and-zero #40: Stereo Card device sharing

noexit23, 2019-12-22:
Just added this new, fancy card to my new Raspberry Pi 4 device.  While I'm just a beginner in the realm of Linux and Pi, my intent is to create a more "eco-friendly" media PC for the kids to watch.  What I like about the A-I Stereo Card is the inputs and outputs.  I have the various HDMI streaming dongles attached to the family TV.  Then I run Audio Out from the TV into a Windows PC which the Pi4 is to replace.  The Line In Audio from the PC is then sent back to the Line Out on the PC to a stereo amp.   This way we can watch local media or streaming media and control the volume with one easy remote.   
The trouble I'm having is getting the various apps on the Pi to "share" the A-I Stereo sound card.  I've got the Line in working by enabling the "Capture" button.  On my Switches tab I have "Output Mixer HiFi" checked and "Output Mixer Line Bypass" checked.  However, I'm unable to, for example, play a YouTube video in a browser window while also playing a local video file via VLC Media Player.  
Is there a way to enable the SoundCard to be shared across all apps or prevent (like in Windows) applications from taking exclusive control of the SoundCard?

flatmax, 2020-01-11:
Hi there,

You need to use the dmix ALSA plugin which allows you to share and 
output device.

You can read the orfficial documentation here : 
https://www.alsa-project.org/wiki/Asoundrc#dmix

MAtt

noexit23, 2020-01-14:
Hey Matt!
Thanks for that guidance.  I've been struggling for a couple of days now
trying to figure out how to enable dmix ALSA plugin and just haven't
cracked it yet.  I'm sure it's quite simple, but then again, I've only been
using Linux for a few months now.  I'm learning lots, I suppose.

Do I define the dmix plugin within .asoundrc using the Code listed on the
Documentation site?

Thanks for your help!

Michael
Johnson City, TN USA

noexit23, 2020-01-17:
Figured it out:
To enable the Dmix Plugin as for my purposes:
Edit the file **/etc/asound.conf**
Add the following text, save, reboot:
**pcm.!default {
        type plug
        slave.pcm "dmix"
}**

ALSO - I noticed when playing back from Squeezelite that Squeezelite's playback was playing too fast.  Following the advice I received here:  https://www.alsa-project.org/main/index.php/Asoundrc

> > Some notes:
> 
> The dmix PCM name is already defined in the global configuration file /usr/share/alsa/alsa.conf.
> 
> - The default sample rate for this device is 48000Hz. If you would like to change it use:
> 
> aplay -D"plug:'dmix:RATE=44100'" test.wav

I changed the following line in **/usr/share/alsa/alsa.conf**
to now read **defaults.pcm.dmix.rate 41000** It was, by default, **48000**.

Thanks Matt and community for your help.