Audio-Injector/stereo-and-zero #49: Reusing audioinjector device tree overlay for custom hardware

jbaker0428, 2020-02-10:
I'm currently migrating a project from a Pi Zero with an AudioInjector Zero to a custom carrier board for a Pi Compute Module.  I'm reusing the WM8731 codec IC and hoping to not have to do any new driver configuration. There doesn't seem to be a schematic posted so I'm running a little bit blind, but I'm only using the mic input so it's not too big of a deal. My bigger concern is hardware changes that might break the dtoverlay driver configuration. 

 * Are you running in I2C or SPI mode? I'm planning on using I2C1.
 * For the clock source, I assume that I should stick to a 12.0 MHz source instead of 12.288 or 14.832. Am I correct in my reading of the datasheet that I should be able to swap the crystal for a MEMS oscillator without issue? Trying to cut board area.

abirkmanis, 2020-11-26:
Disclaimer: I am just another user.
AudioInjector Zero seems to hard-code WM8731 in I2C mode. The chip itself supports both modes (calling them 2-wire and 3-wire).

flatmax, 2020-12-10:
This sort of information is best found in the device tree overlay for the sound card. This is the section which defines I2C behaviour :
https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/audioinjector-wm8731-audio-overlay.dts#L16-L29

That ALSA machine driver also has information you need. For example, this is where the 12MHz clock is setup :
https://github.com/raspberrypi/linux/blob/rpi-5.4.y/sound/soc/bcm/audioinjector-pi-soundcard.c#L82-L85

If you want to, then you can desolder and replace the crystal for one you want to use and change the code there. It should just work !