Audio-Injector/stereo-and-zero #50: No output and no input

robert5974, 2020-02-12:
I've set up this card several times on different versions of Raspbian with the instructions.  The card is recognized as it should be and I can't find anything out of sorts...except there is no sound and the mic isn't producing any input.  The script to test won't run.  It gets to 66% and fails to produce. What could it be?

flatmax, 2020-02-12:
Doesn't sound good.

Is the alsamixer setup to enable audio output ?

You can check here for an example :

http://www.flatmax.org/phpbb/viewtopic.php?f=5&t=5

p.s. there is a new email list up and running : 
https://lists.audioinjector.net/mailman/listinfo/people

robert5974, 2020-02-12:
Thanks Matt.  Yes, the alsamixer is setup just like that.  My goal is to use this in my car so right now I'm setting it up in the house and I have the Rout and Lout using RCA to 3.5mm jack to my Bose speaker AUX input.   Do you see a problem doing that to test?

robert5974, 2020-02-12:
I'll post in a few minutes my settings for you to see.  Thanks.

robert5974, 2020-02-12:
$aplay -lL
default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=audioinjectorpi
    audioinjector-pi-soundcard, AudioInjector audio wm8731-hifi-0
    Default Audio Device
dmix:CARD=audioinjectorpi,DEV=0
    audioinjector-pi-soundcard, AudioInjector audio wm8731-hifi-0
    Direct sample mixing device
dsnoop:CARD=audioinjectorpi,DEV=0
    audioinjector-pi-soundcard, AudioInjector audio wm8731-hifi-0
    Direct sample snooping device
hw:CARD=audioinjectorpi,DEV=0
    audioinjector-pi-soundcard, AudioInjector audio wm8731-hifi-0
    Direct hardware device without any conversions
plughw:CARD=audioinjectorpi,DEV=0
    audioinjector-pi-soundcard, AudioInjector audio wm8731-hifi-0
    Hardware device with all software conversions
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectorpi [audioinjector-pi-soundcard], device 0: AudioInjector au                                                                                                                                                             dio wm8731-hifi-0 [AudioInjector audio wm8731-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

 $ cat /proc/asound/cards
 0 [audioinjectorpi]: audioinjector-p - audioinjector-pi-soundcard
                      audioinjector-pi-soundcard

$ amixer contents
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=63069,63069
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=58982,58982

aplay /usr/share/sounds/alsa/Side_Left.wav
Playing WAVE '/usr/share/sounds/alsa/Side_Left.wav' : Signed 16 bit Little Endia                                                                                                                                                             n, Rate 48000 Hz, Mono
^CAborted by signal Interrupt...
aplay: pcm_write:2011: write error: Interrupted system call

(killed it because there wasn't any sound produced)

[image]

flatmax, 2020-02-12:
This discussion has shifted to the community email list : https://lists.audioinjector.net/mailman/listinfo/people

https://lists.audioinjector.net/mailman/private/people/2020-February/000001.html