Audio-Injector/stereo-and-zero #52: Electret mic questions

jatahan, 2020-02-21:
Hi there,

Following up from an earlier email, can anyone tell me what the bias voltage for the electret mic is?

And I think the unit ships with an electret capsule, what's the part number?

Thanks,
Jem

Srivallabha-k, 2020-03-10:
Hello, 

Does anyone know the model number of the Electret microphone which is supplied with Zero soundcard?

Thanks
KS

flatmax, 2020-03-11:
Similar specs to the WM-52B electret microphone.

Srivallabha-k, 2020-03-11:
Thanks for your response, Matt. But, can you point me to the exact microphone which is in use, that would really help me?

flatmax, 2020-03-11:
The model is 52D electret microphone.

it is similar to the wm-52b microphone

Srivallabha-k, 2020-03-11:
Hello Matt,

Really appreciate your response. I am facing an issue with Zero soundcard, randomly we get to hear blips/pops in the recorded audio files. We noticed this happening in two different samples, Please listen to attached audio sample at around 2 seconds you will hear a pop. It is more frequent in one hardware over another. Is this a known issue? How do we fix this?  If it helps i am recording audio at 44.1khz, S16_LE format, wav format with rest of parameters being the default.  Your response would be highly appreciated. 

Thank you.

[2020_03_11-15_40_16.zip](https://github.com/Audio-Injector/stereo-and-zero/files/4321410/2020_03_11-15_40_16.zip)

flatmax, 2020-03-23:
Are the pops to do with overruns ? When ALSA misses capturing samples 
because the buffer is full ?
Do you see anything reported about xruns ?

Srivallabha-k, 2020-03-24:
I don't think it is something to do with buffer. I tried varying buffer size and that didn't have any effect. Also, if it has to do with buffers the frequency of occurrence will be precise which isn't the case.  As reported in #53, it is not dependent on buffer size. As per our tests on two different setups, It also slightly varies from one hardware to another in terms of frequency of occurrence,

jatahan, 2020-03-25:
I also have the pops on recording and my case it's interference from the wifi. When I have the AI board mounted directly over an Zero W (i.e. gpio pins aligned, 10mm space between boards) I get pops on recording that align with wifi transmission. To test, try recording while pinging the pi at 1s intervals with no other traffic on the interface and you should get pops separated by 1s on your recording.  I haven't got around to testing with different alignments yet.

flatmax, 2020-04-02:
Oh yep - a good reason to isolate the sound card. There is likely currents being induced by EMI, which  sullies the ground plane.