Audio-Injector/stereo-and-zero #54: Zero Power Pins

w4mmp, 2020-03-03:
Hello,
We are using the zero card as part of a larger project.   Currently I have the zero card attached to the RPi 4 with Dupont jumpers so that I can connect an I2C device to an alternate I2C buss.   (This will not be the finial product configuration.  This is only for prototyping.   The final configuration will be with the audio injector hat placed on a connector with extra long pins with that a 40 pin ribbon cable may be attached on top of the zero audio injector).
At any rate, I need to know the power pins for the Zero card.  I found documentation regarding the signalling connections needed but I can find the configuration of the power pins.
NOTE:  Even without the zero connected to a power source,  aplay -l does report the zero is attached to the RPi 4.

flatmax, 2020-03-03:
The zero uses both the 5V pins and the 3.3V power pins. It also uses the 
GND pins. It would be good practice to wire all of those pins in.

w4mmp, 2020-03-03:
Hi,

Thanks for the info!  It helps a lot.

Ron