Audio-Injector/stereo-and-zero #56: arecord overruns

molacour, 2020-03-26:
Hello,

I have a Raspberry PI ZW with a Zero soundcard, I would like to record from the RCA input using arecord but I can't get a proper recording without overruns.

I tried to use both hw:0,0 and plughw:0,0; increased the niceness to -19, reduced the running services to the minimum (no X, no HDMI, no WLAN), and even adding the dtoverlay = i2s-mmap as mentioned in issue "I2S SYNC error #15" but nothing seemed to improve the issue.

The only thing that helped was to record to a USB Drive formatted as XFS.

Here is an example log:
$ arecord -v -f dat -t wav -D plughw:0,0 /media/usb1/20200326140341.wav
Recording WAVE '/media/usb1/20200326140341.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Plug PCM: Hardware PCM card 0 'audioinjector-pi-soundcard' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
  appl_ptr     : 0
  hw_ptr       : 0
overrun!!! (at least 3.829 ms long)
Status:
  state       : XRUN
  trigger_time: 81.478457171
  tstamp      : 0.000000
  delay       : 0
  avail       : 24000
  avail_max   : 24000
overrun!!! (at least 137.395 ms long)
Status:
  state       : XRUN
  trigger_time: 111.474394625
  tstamp      : 0.000000
  delay       : 0
  avail       : 24000
  avail_max   : 24000
overrun!!! (at least 80.945 ms long)
Status:
  state       : XRUN
  trigger_time: 202.118112462
  tstamp      : 0.000000
  delay       : 0
  avail       : 24000
  avail_max   : 24000
overrun!!! (at least 152.560 ms long)
Status:
  state       : XRUN
  trigger_time: 407.572978308
  tstamp      : 0.000000
  delay       : 0
  avail       : 24000
  avail_max   : 24000
overrun!!! (at least 20.600 ms long)
Status:
  state       : XRUN
  trigger_time: 532.723793464
  tstamp      : 0.000000
  delay       : 0
  avail       : 24000
  avail_max   : 24000
overrun!!! (at least 38.344 ms long)
Status:
  state       : XRUN
  trigger_time: 538.869608389
  tstamp      : 0.000000
  delay       : 0
  avail       : 24000
  avail_max   : 24000
overrun!!! (at least 13.069 ms long)
Status:
  state       : XRUN
  trigger_time: 653.144806087
  tstamp      : 0.000000
  delay       : 0
  avail       : 24000
  avail_max   : 24000
Aborted by signal Interrupt...

Have you got any idea about this?
Thanks

flatmax, 2020-04-02:
Ok - check out the buffer duration or period size options for arecord :

-B, --buffer-time=#     buffer duration is # microseconds
     --period-size=#     distance between interrupts is # frames

Try first to play with the buffer time. Choose a long delay - like 100 
ms. Then shorten that to 10ms. Then shorten that to 1ms.

See what works

Matt

molacour, 2020-04-02:
I did not managed to set the --buffer-time parameter alone for some reason but it works with the following command:
arecord -v -r 48000 -f S16_LE -c 2 --buffer-size=48000 --period-size=6000 -D plughw:0,0 /media/usb1/test.wav

I let it run for 40min now without an overrun, so just doubling the buffer-size seems to do the trick for me.
Thanks very much for the help!

closed 2020-04-02 by molacour

flatmax, 2020-04-03:
You may be interested in our latested kickstarter. If you haven't checked it out, follow the kickstarter link from our site : http://audioinjector.net/