Audio-Injector/stereo-and-zero #57: zero soundcard install on arch linux

aidanreilly, 2020-04-17:
I'd like to use a zero sound card with the locus sonus streambox arch linux based distro. Is this possible? 

Stream box Arch Linux details are here: http://locusonus.org/streambox/README.html

http://locusonus.org/streambox/

flatmax, 2020-04-18:
Yes you can - from what I read on their Readme

The setup should be similar as for Raspbian.

You may want to confirm that the overlay is present for loading :

find /boot -name 'audioinjector-wm8731-audio*'

Modify the config.txt file, try first to use this :

#dtparam=audio=on

dtoverlay=audioinjector-wm8731-audio

After rebooting, check to see aplay -l lists the Audio Injector sound card.

=================================

Check out our latest kickstarter :
https://www.kickstarter.com/projects/flatmax/audio-injector-isolated-dac-and-adc?ref=ppNew
Join the community discussion forum :
https://lists.audioinjector.net/mailman/listinfo/people

aidanreilly, 2020-04-18:
Thanks! aplay -l gives me: 

```
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectorpi [audioinjector-pi-soundcard], device 0: AudioInjector audio wm8731-hifi-0 [AudioInjector audio wm8731-hifi-0]
```

Does this look alright? The audio stream is not starting up on the streambox, so i guess there is a configuration somewhere else that also needs t obe set to recognise the zero card.

maybe somthing in the darkice config? /home/locusonus/bin/darkice.sh 

```
* streambox-darkice.service - Locusonus Streambox Live Stream
     Loaded: loaded (/usr/lib/systemd/system/streambox-darkice.service; enabled; vendor preset: disabled)
     Active: activating (auto-restart) (Result: exit-code) since Sat 2020-04-18 08:51:08 UTC; 3s ago
    Process: 28912 ExecStartPre=/home/locusonus/bin/audio.sh --setup (code=exited, status=0/SUCCESS)
    Process: 28922 ExecStart=/home/locusonus/bin/darkice.sh (code=exited, status=255/EXCEPTION)
    Process: 28923 ExecStartPost=/usr/bin/sleep 7 (code=exited, status=0/SUCCESS)
   Main PID: 28922 (code=exited, status=255/EXCEPTION)
```

aidanreilly, 2020-04-18:
OK, just needed to set soundcard location in the streambox config. all good i think! 

`SOUNDCARD=0`

aidanreilly, 2020-04-18:
I'd like to use the onboard electret mic by dafault. Is there a way to set this as the default input?

thanks :)

aidanreilly, 2020-04-19:
Ok I just needed to use ALSAmixer. All good.

closed 2020-04-19 by aidanreilly

flatmax, 2020-04-23:
Yes, you should work out how to store the amixer state. First setup the 
state you want to keep then store it.

alsactl store

Should do the trick.

You may need to alsactl restore at boot to get the state back - normally 
it is autocalled for you, but not sure about the setup which is in place 
in your situation.

aidanreilly, 2020-04-24:
Hi, I was albe to persist settings across power up/down with `sudo alsactl store` - I think. took the device apart to add 40 ribbon connector so it's not running currently.