Audio-Injector/stereo-and-zero #58: Loud WiFi noise using onboard electret - zero board

aidanreilly, 2020-04-19:
I have 40 pin soldered on a rPi zero and AI zero board, with the electret soldered in place on board. I’m getting very bad WiFi interference. Periodic noise pops, quite loud. I’m trying to use this for icecast broadcast. WiFi is absolutely required. I’ve cycled through the channels on the router as described here with no real difference - periodic noise does not go away: http://forum.audioinjector.net/viewtopic.php?t=3133

Any ideas? Is it possible this is some other issue?

aidanreilly, 2020-04-23:
Radio silence on this question :/ planning to mount board away from pi as soon as I get 40 pin ribbon. 
 
Is this a limitation of the zero board - not suitable for streaming broadcast application? If it is, it should be documented up front.

aidanreilly, 2020-04-23:
Here is an example of similar products that are well documented. Please consider a similar document for future IA products.

aidanreilly, 2020-04-23:
http://www.iqaudio.com/downloads/IQaudIO.pdf

dmooney65, 2020-04-23:
Thi doesn't sound like Wifi interference to me. I wrote the comments you refer to and I was not getting any noise. I just couldn't play HD audio from a shared drive over wifi reliably unless I switched channels.

How are you hearing the pops, through headphones or speakers? If the latter then it's probably feedback.
To narrow down, try connecting headphones to the output. If the pops etc. are still there then it isn't feedback, so you could move on to disabling wifi.
Also try simply lowering the input volume.

Maybe you need to use a proper microphone and not an electret.

aidanreilly, 2020-04-23:
Thanks for reply :) 

Issue is experienced on the output end of an icecast ogg stream listening on isolated headphones. “Local” testing with recording with electret and playback on headphones did not have an issue.

dmooney65, 2020-04-23:
OK, so definitely a wifi issue then! Probably best to include what you've tested so far up front for future reference though ;)

I found that a ribbon cable does help a fair bit, but you may need to play aroud with some sort of shielding around the audioinjector too. All sorts of stuff will act as a shield.

Maybe try different sample rates too?

aidanreilly, 2020-04-23:
Yup thanks. Will experiment and report back here.

aidanreilly, 2020-04-28:
ok, mounting with a 40pin ribbon cable doesn't seem to do much. if i hold the AI board at a perpendicular angle to rpi the noise disappears from the ogg stream. seems like this little card is not going to work out for me. what else can i do? will mounting the mic off board help any?

aidanreilly, 2020-04-30:
OK, mounted the electret offboard with some running wires and didn't notice any improvment. I'm going to give up on this use streaming use case :(

flatmax, 2020-06-25:
Thanks for letting us know. I wasn't aware of this problem.