Audio-Injector/stereo-and-zero #60: Audio Injector Soundcard and GPIO ribbon

smomid, 2020-07-01:
Hi, 

I have a R Pi 4. I need to have all my i2cd devices connected on a GPIO ribbon connected to the Pi. When I connect the Audio Injector to a GPIO ribbon the Pi doesn't boot up. I tried a few other devices and they work, just not the Audio Injector. 

Is there a way to make this work, either connected to the pins of another i2c device, or to a GPIO Ribbon. 

I am also using a Pi juice and a LCD screen shield. Thanks, Nick

flatmax, 2020-07-01:
I always prototype using the quick connect clip+ribbon :

https://shop.audioinjector.net/detail/DIY_Electronics/PCB+quick+connect+clip

Is there any way to show an image of your connected system ?

Polarity matters. Is it possible that you have one of the board around 
180 degrees the wrong way ?

This has been known to let the smoke out !

smomid, 2020-07-03:
Hi. I'm pretty sure it's not how I have the boards connected. I did a bunch of tests today. I think there is some conflict on the i2c bus with the Pijuice and Audio Injector. 

I got the Audio Injector to work on a GPIO ribbon, and I got the Pijuice to work on a GPIO ribbon. If I connected the PiJuice, then the Audio Injector the Pi won't boot up all the way. If I connect the Audio Injector then the PiJuice the PiJuice is not appearing to be connected. 

The Pijuice has some configurable pins on the i2c. i2c address one is 14, i2caddress2(RTC) is 68. ID EEPROM address is 50 (but can be changed to 52). I2c address 1 and 2 can be changed. Is it possible that this is contributing to the issue? Thanks

smomid, 2020-07-07:
Hi, I wanted to know if anyone had a chance to look at my response above and if there is a conflict in i2c addresses. 

thanks

flatmax, 2020-07-07:
This is the i2c setup for the audio injector stereo-zero product :

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/audioinjector-wm8731-audio-overlay.dts#L22

Is there a register clash on the 1a address with the pijuice ?

smomid, 2020-07-07:
On the Pijuice address 1 is 14 and I don't see that in that code, but then again I don't know much about i2c. 

I can try changing address 1a on the Pijuice to something else and see if that works.

smomid, 2020-07-19:
hey flatmax. What is a safe number to change the i2c address 1 of the PiJuice to? Any suggestions? Thanks.

flatmax, 2020-07-19:
Interesting that the piJuice can change its i2c address. Not sure what 
an unused address is on your system - most likely everything except 1a !

You can probe what is present on i2c using the i2cdetect command. I 
think it is bus 1 :

i2cdetect 1

It will output "--" for available addresses. You can most likely choose 
any of those available addresses.