Audio-Injector/stereo-and-zero #61: having issues with i2c....

smomid, 2020-08-16:
Hi, 

I unplugged my Audio Injector Stereo because I needed to use the GPIO pins for something else. 

I put the Audio Injector back on, but I think I forgot something or messed something up. 

What i2c addresses does the Audio Injector use? 

Once I change /boot/config.txt to have the audio injector selected and the default audio off, is there anything else I need to do? 

With the situation I have now, sometimes I run aplay -l and the audio injector shows up, but no sound comes out. I think I jsut have something messed up, or there is a conflicting issue with i2c addresses. 

[image]
[image]

smomid, 2020-08-18:
[image]
[image]
[image]
Hey i am still trouble shooting this issue and I wanted to submit these images, maybe they provide some clues. 

I tried unplugging everything from the GPIO pins except the Audio Injector and if I did "aplay -l" in the terminal it said no sound card is present. 

Once it said Audio Injector was present, but no sound came out of it. I had the volume turned up too. I must have some kind of conflict with something on the i2s or i2c bus, even though nothing is plugged into it. It is also plugged in the correct way. 

Let me know what you think I can do to resolve this issue. Thanks. Nick

flatmax, 2020-08-18:
Yes, that definately looks like problems communicating with the audio injector sound card at a driver level.

The I2S sync reports are not a problem.

However your later images clearly show a failure to load the sound card.
If you disable the sound card in the config.txt file and fun "i2cdetect 1" do you see the sound card in the matrix printed out ?

smomid, 2020-08-18:
HI Matt, 

II disabled audio injector in the config.txt file, rebooted. I have no other things plugged into the GPIO

I ran i2cdetect 1 and I see something on in the Matrix at 10 a that says 1a. I’m guessing that’s the Audio Injector. 

What do you think? 

Nick

flatmax, 2020-08-18:
yes, that is its address.

You should be able to start it up now and see the sound card loads ?

Matt

smomid, 2020-08-18:
Matt, 

Thanks for the reply. 

So I uncommented the Audio Injector in the config.txt file and rebooted. 

I am still getting a ton of errors when I run dmesg in the terminal. 

When I run aplay -l in the terminal is says no sound cards are detected. 

Then I ran i2cdetect 1 again and now, at address 10 a in the matrix is says UU where it said 1A before. 

What do you think? I must have turned something off or something. I looked and i2c and i2s are active in the config.txt

Thanks, Nick

flatmax, 2020-08-18:
OK,

The UU means that it is reserved by a Linux driver - however you know 
the sound card is the one and you also know there is no clash for 
whatever reason.

Are there any eletrical anomalies on the board ? Did the smoke come out 
of anything at any point that you are aware of ?

smomid, 2020-08-18:
No, no smoke came out. I think everything is good. I’ve been careful when plugging it in or unplugging it. Pretty sure it’s plugged in correctly. 

The first time I used the Audio Injector I manually typed everything in the config.txt. 

I think I tried doing the automatic method before manually messing with the config.txt this time and I installed something I downloaded from your GitHub 

I wonder if there is an application from the automatic install method that is reserving that address, like two things for the audio injector are trying to use that address. 

I also have the Pijuice, but I unplugged it, and i don’t think it uses address 1A anyway. I uses 16, 50 and 68. 

Nick

smomid, 2020-08-20:
Hey Matt, 

What was installed in the automatic method of setting up the Audio Injector? 

I think I might just try removing everything, then having Audio Injector selected in the config.txt file. maybe that will solve the issue. 

thanks, Nick

flatmax, 2020-08-20:
Here is the automated setup script :

https://github.com/Audio-Injector/stereo-and-zero/blob/master/audio.injector.scripts-0.1/audioInjector-setup.sh

it does the following :

1] runs rpi-update - this may not be necessary depending on the age of 
your OS

2] Comments out the audio devicetree in /boot/config.txt : #dtparam=audio

3] sets up the dtparam for the sound card : 
dtoverlay=audioinjector-wm8731-audio

smomid, 2020-08-21:
Matt, 

OK. Well I tried doing all that stuff already, so I guess that’s not it. 

Can you think of anything else I need to try? 

Thanks, Nick

flatmax, 2020-08-21:
Do you have a second sound card and Pi to test with ?

smomid, 2020-08-21:
I have another brand new audio injector and I have another PI, but not another SD card with a different disc image. I could put a disc image on a new SD card, but it might take a day or so.

flatmax, 2020-08-21:
I forget what operating system you are using. Could that be the cause of 
the problems ? Did you test with the stock standard Raspbian ?

smomid, 2020-08-21:
I’m using Patchbox OS.

No, I don’t think it’s the OS. I think I accidentally changed some setting somewhere. Or disabled something. When I first set everything up it was working fine, but I unplugged the Audio Injector for a few weeks and plugged some other stuff into the GPIO pins. Something changed because when I went to plug the Audio Injector back in it wasn’t working. 

Nick