Audio-Injector/stereo-and-zero #64: jackd don't receive inputs.

abnervios, 2020-11-10:
Greetings. 

I tried the "guitarix" software, an amplifier simulator, works with JACK... I tried routing the audio signal with JACK (I know about JACK works. I've used it for years) but guitarix don't receive any signal. If I connect the guitarix outputs to the system playback inside JACK, I can hear some noises filtered, example: the flanger sound. But not so much.
I tried to connect directly the Ins and Outs with JACK and no sound.

When JACK starts, I can hear a pop.
When I connect any Jack ins to out, I hear a pop.

I can hear the mic and line inputs when the "output mixer mic sidetone" and "output line bypass" are activated in alsamixer.

Greetings and sorry for my bad english.

abnervios, 2020-11-10:
SOLVED!

I didn't set the capture settings in alsamixer. Bye

flatmax, 2020-11-10:
Good going !

Can you send a capture of your mixer settings here for others in the 
future ?