Audio-Injector/stereo-and-zero #66: Application reading 12c device fails [No I2C device at address: 5a]

pankajkgupta, 2020-11-18:
Hi, before connecting audioinjector stereo to my Pi 4B, I was able to use adafruit breakout board that reads i2c at address 5a.
But after plugging the audio injector, and connecting the GPIO ribbon, i2c read seems to fail.

I ran `i2cdetect 1` and following output was shown:
`pi@raspberrypi:~ $ i2cdetect -y 1`
`     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f`
`00:          -- -- -- -- -- -- -- -- -- -- -- -- -- `
`10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- -- `
`20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- `
`30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- `
`40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- `
`50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- `
`60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- `
`70: -- -- -- -- -- -- -- --                         `

I am not sure how to get i2c applications going with the audioinjector GPIO pin extensions.

Any help[ would be appreciated, otherwise I will have to set the audioinjector aside.

Pankaj

flatmax, 2020-11-18:
You seem to still be loaing the audioinjector overlay (in config.txt). 
Address 1a is the audio injector stereo soundcard.

Do your other I2C devices also use the 1a address ?

pankajkgupta, 2020-11-24:
@flatmax thanks for your reply.
My other device uses address 5a.

> You seem to still be loaing the audioinjector overlay (in config.txt).

Should I not be loading audioinjector overlay? What can I try to avoid doing this?

Thanks!
Pankaj

flatmax, 2020-12-10:
It seems that 5a is empty and unused. You should be able to load the AI overlay without effecting your i2c communictions on address 5a. They should all be able to share the same i2c bus.

Have you cross checked that none of the other GPIO lines in use are conflicting ?