Audio-Injector/stereo-and-zero #70: Soundcard recognized but no output

SallarShayegan, 2020-12-30:
Hello, I have connected the soundcard with my raspberry pi 4 and have run the installation script from this repository. The card is listed when I use aplay -l command, but there is no sound output. Btw I have already turned off the default sound output and any other sound devices, so audioInjector soundcard zero is the only (default) sound card in my system.

here is a screenshot of alsamixer of the soundcard on my rpi
[https://ibb.co/kKNvwdq](url) 

also when I run the test script I get this:
`play: no process found
arecord: no process found
gpicview: no process found
Recording WAVE '/tmp/test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
play WARN alsa: can't encode 0-bit Unknown or not applicable

 File Size: 0         
  Encoding: n/a           
  Channels: 2 @ 16-bit   
Samplerate: 48000Hz      
Replaygain: off         
  Duration: unknown      

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    arecord: pcm_read:2145: read error: Input/output error
sox WARN wav: Premature EOF on .wav input file
Recording WAVE '/tmp/test.mic.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
play FAIL sox: Sorry, there is no default audio device configured
play WARN alsa: Input/output error
play FAIL sox: `default' Device or resource busy: Operation not permitted
Done.
arecord: pcm_read:2145: read error: Input/output error
sox WARN wav: Premature EOF on .wav input file
press enter to test again, any other key then enter to exit
`
in puredata and other softwares I keep receiving "device is busy" message reffering to the card
By the way I have not installed the microphone and rca in- and output slots on the board, because I am going to use different input.

what can I do to fix this?

Thanks in advance

-Sallar Shayegan

SallarShayegan, 2021-01-12:
@flatmax I kindly remind you to please have a look on this. It has been about a month since I've received the soundcard but unfortunately I have still no clue how to make it work

flatmax, 2021-03-18:
I would like to remind github users, that we don't officially support here - search for audioinjector people. Use our email list.

Can you show the output of aplay -l ?
Is pulseaudio installed ?

jmsiener, 2021-03-19:
@SallarShayegan I ran into similar issues as you did and I wasn't having much luck with the provided scripts either.  I ended up doing what's documented here: https://github.com/mxmxmx/terminal_tedium/wiki/setting-up-linux .  It's for a eurorack module which uses the same codec (WM8731) and this process has gotten my hat working but I haven't tested the electret mic yet.  I don't really have much application for built in mic anyway since I have access to mics+pres and can just use the line input.  It's working fine even with pulseaudio installed but you do have to go into your specific soundcard config in alsamixer (F6) and enable an option; the "m" key of your keyboard will toggle these on off, don't ask me why they didn't use enter or space or anything else but I suppose it could imply mute.  Hope this helps.