Audio-Injector/stereo-and-zero #71: AudioInjector no longer recognised by Pi

nikloynes, 2021-02-20:
Hi there.

I've been using the AudioInjector Stereo card for a while with my RPi3, with the primary purpose of running a samplerbox instrument on it (https://homspace.nl/samplerbox/SBabout.html). Back in May 2020 or so, I managed to get it working and configured (which involved more effort than just running the install instructions detailed on the github page, such as configuring the default soundcard - but I can't quite remember what exactly it involved).

Anyway, I didn't touch the samplerbox-pi for ages, and only last week wanted to incorporate it into my setup again. However, despite making zero changes to the configuration and not touching the thing for over half a year, it stopped working. I wanted to start debugging it by simply getting the pi to play a test wav file by simply running `aplay test.wav`, but this results in an error message telling me no soundcard was found. 

So, I thought I'd re-install samplerbox with a clean image - but the same thing happened. Then I thought, perhaps I should test Raspbian / Raspberry Pi OS - same issue. See below for examples: 

```
pi@raspberrypi:~ $ aplay -l
aplay: device_list:272: no soundcards found...
pi@raspberrypi:~ $ alsamixer
cannot open mixer: No such file or directory
pi@raspberrypi:~ $ arecord -l
arecord: device_list:272: no soundcards found...
```

My config.txt looks as it should, with the on-board sound deactivated and the audio injector overlay activated: 
```
# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable the AudioInjector.net sound card
	dtoverlay=audioinjector-wm8731-audio
```

So, my theory at this point is that perhaps something is damaged on the AudioInjector hat itself? I took it off the Pi and placed it on top again, to no avail. So, I'm attaching some photos of the board. Please note that the imperfect alignment of the RCA jacks was how I received the audio injector when I bought it (new!), and I absolutely didn't touch the Pi for months, so I can't really see how anything could have been damaged. 

[image]
[image]
[image]
[image]

I wonder if you can identify what the issue might be? Is there some damage to the pcb? Or is it software related? Any help will be greatly appreciated, thanks. 

All the best,
Nik

nikloynes, 2021-02-22:
some additional info: 

this is what `dmesg` spits out relating to the AudioInjector:
```
[    6.994830] audioinjector-stereo soc:sound: ASoC: failed to init link AudioInjector audio: -517
[    6.994852] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    7.119065] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    7.248955] audioinjector-stereo soc:sound: ASoC: failed to init link AudioInjector audio: -517
[    7.248980] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    7.253906] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    7.262051] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    7.262379] usbcore: registered new interface driver brcmfmac
[    7.264222] audioinjector-stereo soc:sound: ASoC: failed to init link AudioInjector audio: -517
[    7.264242] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
```

and
```
[    8.009857] wm8731 1-001a: Assuming static MCLK
[    8.009903] wm8731 1-001a: 1-001a supply AVDD not found, using dummy regulator
[    8.010021] wm8731 1-001a: 1-001a supply HPVDD not found, using dummy regulator
[    8.010114] wm8731 1-001a: 1-001a supply DCVDD not found, using dummy regulator
[    8.010202] wm8731 1-001a: 1-001a supply DBVDD not found, using dummy regulator
[    8.123022] wm8731 1-001a: Failed to issue reset: -121
[    8.124779] wm8731: probe of 1-001a failed with error -121
```

BoxOfSnoo, 2021-02-27:
You should probably post this (or a link to this) on the mailing list.  I think they migrated away from GitHub issues a while ago.

flatmax, 2021-03-03:
This discussion has moved here to the people mailing list :
https://lists.audioinjector.net/pipermail/people/2021-March/000228.html