Audio-Injector/stereo-and-zero #72: Unable to record using "input" mic

nonfatalexec, 2021-04-11:
I'm trying to get the microphone working. I tried connecting the mic to either of the 2 input ports on the right side, but there's no difference. I'm able to record using the "built-in" mic you see next to the input cable. The arecord output shows only the "built-in" mic. What am I missing?

[image]

`pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: audioinjectorpi [audioinjector-pi-soundcard], device 0: AudioInjector audio wm8731-hifi-0 [AudioInjector audio wm8731-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

[image]

[image]

flatmax, 2021-11-23:
can you hit the email list on this topic ?
https://lists.audioinjector.net/mailman/listinfo/people

cabasse, 2021-11-25:
that cable you have plugged into the input breakout board looks like a 3.5mm/TRS type cable, not RCA - is that correct?

nonfatalexec, 2021-11-25:
I think it was a 3.5mm/TRS type cable. It was definitely not an RCA cable.