Audio-Injector/stereo-and-zero #73: Manual setup failing on Raspberry Pi 4

seanjean3000, 2021-05-08:
I can't seem to get my AudioInjector working after many attempts following the manual setup instructions. I looked through other issues here but didn't see this one.

I have a fresh install of Raspberry Pi OS (2021-03-04, buster armhf). After starting up the Pi for the first time with the new OS, I:
1) ran the setup wizard and updated the Pi
2) ran rpi-update
3) edited boot/config.txt to comment out the onboard audio device and add the overlay for the audioInjector
4) powered down, connected the audioInjector and rebooted
5) Downloaded the setup files, put asound.state.RCA.thru.test in /usr/share/doc/audioInjector/

When I run `alsactl --file /usr/share/doc/audioInjector/asound.state.MIC.thru.test restore` per the setup instructions, I get the message: 
```
No state is present for card CinemaTM
Found hardware: "USB-Audio" "USB Mixer" "USB045e:075d" "" ""
Hardware is initialized using a generic method
No state is present for card CinemaTM
```
And, not surprisingly, when I run audioInjector-test.sh it fails:
`play: no process found
arecord: no process found
No state is present for card CinemaTM
Found hardware: "USB-Audio" "USB Mixer" "USB045e:075d" "" ""
Hardware is initialized using a generic method
No state is present for card CinemaTM
Recording WAVE '/tmp/test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
play WARN alsa: can't encode 0-bit Unknown or not applicable

 File Size: 0         
  Encoding: n/a           
  Channels: 2 @ 16-bit   
Samplerate: 48000Hz      
Replaygain: off         
  Duration: unknown      

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    arecord: pcm_read:2145: read error: Input/output error
sox WARN wav: Premature EOF on .wav input file
alsactl: load_state:1683: Cannot open /usr/share/doc/audioInjector/asound.state.MIC.thru.test for reading: No such file or directory
Found hardware: "audioinjector-p" "" "" "" ""
Hardware is initialized using a generic method
Found hardware: "USB-Audio" "USB Mixer" "USB045e:075d" "" ""
Hardware is initialized using a generic method
Recording WAVE '/tmp/test.mic.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
play FAIL sox: Sorry, there is no default audio device configured
In:0.00% 00:00:02.65 [00:00:00.00] Out:127k  [ =====|===== ] Hd:5.9 Clip:0    press enter to test again, any other key then enter to exit
In:0.00% 00:00:03.07 [00:00:00.00] Out:144k  [ =====|===== ] Hd:5.9 Clip:0    
Done.`

Any help would be appreciated.

tnesseth1, 2021-06-06:
me as well

the pi 4 with the octo channel board connected won’t boot. it will boot if i remove the hardware -?? - it my might be a pin thing from what i’ve read? any thoughts?

flatmax, 2021-06-06:
It isn't finding your card perhaps ?
aplay -l
What does that list ?

seanjean3000, 2021-06-16:
Well, I had put it all away since I posted that, not having made any progress. I took it all out again last night to see where I can get.

aplay -l definitely lists the device (and always did).

I was able to get pass-thru of audio (line signal from the inputs to the outputs) which *sort of* responds to the line bypass in alsamixer, which suggests to me that the AudioInjector is working, nominally. But I say "sort of", because if I toggle the line bypass on and off, the audio doesn't always come back through--sometimes I have to toggle it several times.

alsactl --file /usr/share/doc/audioInjector/asound.state.MIC.thru.test restore and audioInjector-test.sh still fail (I don't have the device near me so I can't give the exact error messages, but for the former, I believe it is something like "can't make these changes").

Jack sometimes but rarely works. When I launch jackd from the command line with device 0,0 there are no error messages, so jack seems to be recognizing the device, and inputs and outputs are shown in Patchage (which I use for easier visual trouble shooting). Every few times I launch jack though I get a sync error message of some sort and I'm told the server has stopped (also hard to share the exact message because it doesn't always happen). Even then, without error messages / jack apparently running, I don't usually get any audio output from jack clients...

Very rarely, I am able to get sound from a test patch in Pure Data out, using jack, but not always or even usually, and I have no idea why it works when it does. I haven't tried using alsa for Pure Data, but ultimately my need for this project is jack.

So, overall, very unreliable and I have no idea why. This is otherwise a clean install of the latest Raspberry Pi OS.

flatmax, 2021-06-16:
OK,
How about pulseaudio ? Is that taking the default device ?
Can you give a test by altering your .asoundrc. This will override pulse as the default device, create the file ~/.asoundrc and put the following in it :

```
@hooks [
	{
		func load
		files [
			"~/.asoundrc"
		]
		errors false
	}
]

pcm.!default {
  type hw
  card 0
}

ctl.!default {
  type hw
  card 0
}

```

seanjean3000, 2021-06-17:
First, thanks for taking the time to check in on this. Much appreciated.

I forgot to mention in my first post, but removing pulseaudio was one of the first things I did initially as part of the setup. Otherwise, it was stock Raspberry Pi OS (latest as of 3-4 weeks ago), rpi-update, changes to config.txt, and then installing jack, pure data, and patchage. That is all.

I cannot try the asoundrc suggestion tonight, but will try as soon as I can. Again, thanks for that.

seanjean3000, 2021-06-18:
So I couldn't find an asoundrc file in any of the usual places on this install--pretty sure there was one somewhere in an earlier attempt with this same Pi and the audio injector, so not sure why not now. I don't know when that file is usually created, if it should be there by default, or if it might have been deleted when I got rid of Pulse. Really not sure...

Anyway, I tried to create a new one at ~/etc/ and copied in the code you provided, but it didn't do anything for me.

If I were to start from scratch with a fresh install, can you recommend any different approach? The auto install definitely doesn't work, but also the manual method doesn't seem to work reliably either (on a Pi 4 at least). I saw in a mailing list thread somewhere that the Pi 4 specifically might have some weirdness, in that it creates a default alsa device that might interfere with the audio injector. Could this be it?

Anyway, yeah, would be great to have a direct line from fresh install to running audio interface.

Thanks again.