Audio-Injector/stereo-and-zero #75: Volume control range problems

markwinger, 2021-11-15:
I am using the stereo audioinjector with OpenAuto pro.  (I originally started with a usb audio if and changed to the injector)

After getting it all I chose to use the rca outputs and got it all working but volume control did not work.  I tried for days to get softvol to work but was unable to.  So I changed to use the headphone output.  This worked but the quite right.

At full volume things are good, clean and loud as expected.  But as I reduce the volume control, the sound decreases way too quickly.   It is completely silent by 60%.  I found some discussion about 'base' volume but could not determine how to adjust this.  I tried changing volume=merge to volume=ignor but this seemed to have no effect.

Any help appreciated........

differentieel, 2021-11-15:
did you lift the Alsamixer volume slider to 90%?

markwinger, 2021-11-15:
I assume you mean the alsa master.

I’m not clear how this works.  In open auto I see output 2 output devices, one called LADSPA Plugin EQ on Built-in Audio Stereo, and another with (echo controlled with…).  The first is set to 100.  I can see the vu meters show the signal passing through the first one.  If I adjust the volume using this it will go louder or softer, but the alsa mixer master moves with it.  The problem is (I think), the alsa mixer lever will be 0% for any level in the pulse audio mixer below 45%.  The alsa mixer lever moves to 1% when pavu is up to 61%.

I believe for this wot work well, the level of the pavucontrol and alsa mixer should match.  Right???

differentieel, 2021-11-16:
http://forum.audioinjector.net/viewtopic.php@f=5&t=3.html

markwinger, 2021-11-19:
That thread is very long and I did not see any discussion relating to my problem.  Was it intended to be a suggestion for a fix.

differentieel, 2021-11-19:
in short:

RPi running Raspbian Buster

1) Edit /boot/config.txt
Code:
sudo nano /boot/config.txt

2) Uncomment these lines (remove the #):
Code:
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on

3) Comment out this line (add a # at the start):
Code: 
#dtparam=audio=on

4) Add the following line just after the one you commented out in the previous step:
Code: 
dtoverlay=audioinjector-wm8731-audio

5) Quit and save (ctrl-x, y, enter)

6) Reboot the pi: 

7) Run alsa mixer:
Code: 
alsamixer

8) Scroll across to the one titled 'Output Mixer HiFi' and press 'm' to unmute it. It should now say '00' above it with a green background. Press 'esc' to save and exit.

That should be everything. You can now test that it works with:
Code: 
aplay test.wav

markwinger, 2021-11-23:
The only thing different in my setup is I dont have the line
Dtparam=i2s=on, uncommented.

Will this fix the level problem, or enable me to use softvol?

But I am away for the next week so it will be a while before I can try it.

differentieel, 2021-11-23:
the only person that can answer your question is you.

flatmax, 2021-11-23:
I think this may help for softvol : http://forum.audioinjector.net/viewtopic.php@t=572.html

That is from the old audioinjector forum, which can only be searched in google.

Untested, but by editing your .asoundrc file, use a softvol for default :

```
pcm.!default {
type plug
slave.pcm "RCAVolume"
}

pcm.RCAVolume {
type softvol
slave.pcm "hw:0,0"
control.name "Softmaster"
control.card 0
}
```

You may also need the ctl defined if it gives you troubles :
```
ctl.!default {
        type hw
        card 0
}
```

Finally if you get pulse audio funk, make sure you short circuit the pulseaudio defaults :
```
@hooks [
    {
        func load
        files [
            "~/.asoundrc"
        ]
        errors false
    }
]
```

You can see some of that on my stackover answer here : https://stackoverflow.com/questions/34452150/how-to-run-alsa-application-without-killing-pulseaudio