Audio-Injector/stereo-and-zero #77: Audio Injector Stereo GPIO Used

pscorbett, 2021-12-23:
Hi guys,

I have been looking through the documentation, and I haven't found this referenced anywhere. What GPIO pins are actually used by the Stereo sound card? I am trying to implement this in a project that will use SPI, and some shift registers and multiplexers, and I want to make sure there is no conflict.

My understanding is that I2C uses different GPIO pins than SPI on the RPi, but I couldn't find which ones (in particular for this device). Thank you!

differentieel, 2021-12-24:
a time ago i found this post on the AudioInjector forum by FlatMax:

For reference, this message notes which GPIO pins are used by the Audio Injector (AISC).
The AISC shares the I2C pins for setup and management of the sound card's codec's registers. These are :
pin 3 : i2c data
pin 5: i2c clk

The AISC uses the I2S pins for communicating audio data to and from the Raspberry Pi. Thses pins can not be shared amoungst add ons and hats when the AISC is in use. The I2S lines are the following pins :
pin 12 : Bit clock
pin 35: LR clock
pin 38: Data in
pint 40: Data out

For power, the required pins are :
pin 2 & 4 : 5V
pin 17 : 3.3V
gnd pins : as many as you want to hook up !

These are shown in the following image.

[image]

pscorbett, 2021-12-24:
Thank you, this is extremely helpful! Sorry I hadn't found this myself, I was looking, but I should have thought to comb through the old forum. Everything on it was pointing me towards the github so I was assuming it had been dead for a long time.

I didn't realize it the Audio injector used the I2C pins as well, this is very good to know, and will certainly inform my design decisions. Just to confirm then, UART should be totally free and cause no conflict? You're comment didn't list them, so I'm assuming that them being highlighted on the element14 graphic has no bearing on the audio injector, right? 

It looks like all of the other pins should be easy enough to avoid. Amazing how quickly a 40 pin header can fill up though.

differentieel, 2021-12-24:
pleased to hear the info did help
about your question - i cannot answer that - maybe @flatmax is able to answer that one
have nice holydays 😉

pscorbett, 2021-12-24:
You too! Thank you again!

flatmax, 2021-12-25:
Confirmed, UART is free !

The Zero and stereo cards have the same pinouts : 
https://pinout.xyz/pinout/audio_injector_zero_sound_card