Audio-Injector/stereo-and-zero #78: PCB Footprint available anywhere?

otem, 2022-02-03:
I'm looking to mount the zero sound card next to a pi zero on a PCB I am designing. It would be great to have the footprint available in order to line up the mounting holes for the input/output. Sorry, this isn't an issue with the card or software, I just couldn't find anywhere else to ask. I was unable to register on the forum.

differentieel, 2022-02-03:
maybe the former post https://github.com/Audio-Injector/stereo-and-zero/issues/77 can be of any service?

otem, 2022-02-21:
Thanks for the response but unfortunately that won't be of much help. I was looking for the physical footprint in order to line up the input/output/mic pins to a PCB I am designing.

differentieel, 2022-02-21:
you have to ask @flatmax - but - but - you may have a no for an answer - the DAC's we are talking about are his inventions - so be careful what you ask - - in other words, think twice

otem, 2022-02-21:
I was originally going to try to contact him, but was unable to find any contact info. I'm definitely am not asking for the full pcb or anything, even a dimensional drawing with the pin locations would suffice.  My goal is to mount an audio injector zero onto a pcb I'm designing using pins rather than jst connectors/wires. Thanks again for your response.

flatmax, 2022-02-21:
Hi differentieel,

As per your request, seems reasonable, I have created the template board :

https://github.com/Audio-Injector/ZeroTemplate

The best place for accurate inforamtion is the [email list](https://lists.audioinjector.net/mailman/listinfo/people)

otem, 2022-02-21:
Thank you @flatmax! This is exactly what I needed!