Audio-Injector/stereo-and-zero #81: Currently unavailable on Amazon

syedzayyan, 2022-11-27:


flatmax, 2022-11-27:
hi - can you try the shop ?
https://shop.audioinjector.net/

syedzayyan, 2022-11-28:
the shop never loaded for me. I can access the products page but cannot check it out. It waits with a loading indicator for an eternity. I tried three browsers, with and without VPN only to realise some of the JS was not loading.
[image]

flatmax, 2022-11-28:
can you clear your cache, remove cookies and force refresh ?

flatmax, 2022-11-28:
oh hang on - I get the same problem.

syedzayyan, 2022-11-28:
cleared the site data, removed cookies and forced refresh. same

flatmax, 2022-11-28:
I just did another test - seems it is working now.

syedzayyan, 2022-11-28:
I think I placed an order as I have a Paypal receipt, but I also had this dialogue...

[image]

closed 2022-11-29 by syedzayyan