Audio-Injector/stereo-and-zero #84: Support for Bullseye?

lacojim, 2023-02-22:
I just installed Bullseye on my RPi3 and I cannot get this card/hat to be recognized. I have two, and I have tried them both. I bought both of these a few years ago, but this is my first attempt to actually use one of them. I have commented out the line in /boot/config.txt:
dtparam=audio=on
and added:
dtoverlay=audioinjector-wm8731-audio

aplay -l shows no sound devices. Any ideas or is this newer OS simply not supported?

nls-GitHub, 2023-02-22:
I had the AudioInjector board working on a Pi 3B+ and later switched to a
Pi 4, both with Bullseye.  The one difference that I found with the Pi4
install is that I now have to use sudo with the aplay or arecord commands,
but other than that, there are no issues.  Board shows up in raspi-config
also.  I have disabled the HDMI audio by modifying
"dtoverlay=vc4-kms-v3d,noaudio".

lacojim, 2023-02-23:
Thanks for the reply. I do have it showing up now with #aplay -l, just have not been able to get any sound from it. I will try installing PA and see if that helps. Would prefer Pipewire, but it's such an old version.

nls-GitHub, 2023-02-23:
My application is only for recording, so I never really tested the playback
features.  I do use the headphone output for monitoring the recording.

lacojim, 2023-02-23:
I have installed Pulseaudio and still cannot make any sound even though the device DOES show up in Pulse mixer. I threw in the towel and ordered a 384khz DAC audiophile card. Hopefully this one will work better. Thanks!

flatmax, 2023-02-23:
did you enable the output HIFI mixer ? Assuming you have the card setup as hw:0 you can do the following ...

amixer -Dhw:0 set 'Output Mixer HiFi' 'on'

lacojim, 2023-02-23:
Interesting. I did not do this. Thanks for the tip. I will give this a shot in the morning. Card0 is the built-in HDMI and Card1 is the AudioInjector. So I should be able to use -Dhw:1 instead of hw:0. This is on the Rpi 3B+ BTW. I have a Rpi4 which I may switch to at some point.

nls-GitHub, 2023-02-24:
It should be settable via alsamixer also.  The following are my settings,
but I don't use the board for playback, but only recording.  Some fields
are changed by arrow keys, some others by the 'm' key.

[image: image.png]

lacojim, 2023-02-25:
I installed Volumio OS and followed the instructions for this particular card/hat. Still no sound. Then I found a small snippet that mentioned Pulseaudio needs to be removed. So I did, and voila! It worked! The new card will be here tomorrow (Saturday) so we'll see how this one works as well. It's not an AudioInjector card, but I can certainly use more than one DAC in the household so a nice added addition hopefully.

lacojim, 2023-02-26:
I have two AudioInjectors running Volumio now, and one Inno device. All sound pretty darn good. The three rooms are nothing alike, so nothing scientific. Each sounds different according to the room, speakers and amps involved. My main system and most used is in the garage, a Rpi3 running through a vintage Pioneer SX-780 amp with the Inno card with DYI speakers. The second one is running in the Livingroom with an AudioInjector also on a Rpi3 into a Onkyo TX-NR609 amp with bookshelf speakers plus subwoofer. The third system is also an AudioInjector on Rpi4 running the bedroom speakers. Dayton Audio tube amp. All sound pretty darn good.

All good now. Thank you both for the help and advice. Very much appreciated.

lukexi, 2023-11-06:
> I had the AudioInjector board working on a Pi 3B+ and later switched to a Pi 4, both with Bullseye. The one difference that I found with the Pi4 install is that I now have to use sudo with the aplay [...]

To fix needing sudo, you can add the pi user to the audio group with ```sudo adduser pi audio```. Log out and log back in and you should be able to use `aplay` and `arecord` without sudo!

(also can confirm audioinjector is working great here on Bullseye / kernel 6+ on a Pi Zero W original)